Loc15/PicoGamepadConverter #4: Improveming wording/layout for web page

andrewfstratton, 2024-04-08:
I found the web page confusing when first using it.  Could this following help (new) users:

1. Put an explanation that Host is the plugged in real gamepad and Device is how the Pico looks to a Computer
  - possibly give an/some example/s
  - putting the Device setting in (further) features would simplify the typical usage - or just putting it further down the page...
2. Include a link to the github for further detail
3. More detail on 'Block Analogues' I have no idea what these are.

Loc15, 2024-04-09:
> 3. More detail on 'Block Analogues' I have no idea what these are

Block Analogues means to ignore input from gamepad analog stick, it gonna keep in the center.

andrewfstratton, 2024-04-09:
Would 'Ignore Analogue (axis)' be better understood?  I thought Block meant that the analogues work in a 'block' mode, e.g  like analogue to digital...

Loc15, 2024-04-11:
Disable analog stick would be better, I going to adding to documentation on the next update.

closed 2024-06-21 by Loc15