Loc15/PicoGamepadConverter #6: Building instructions

andrewfstratton, 2024-04-15:
Please could the build process be given/checked?  Please reuse anything below if useful...

I have had issues building (after a 35 year lapse in writing C/C++).  So far I have done the following:

1. Installed Docker Desktop (I'm using windows) - following this guide https://community.element14.com/products/raspberry-pi/b/blog/posts/working-with-pi-pico-c-c-projects-easily-a-docker-approach
2. Checked out the project using `git clone https://github.com/Loc15/PicoGamepadConverter.git` into /home/dev/pico
3. Entered these commands - last command gave an **Error for Pico-PIO-USB**

```
mkdir build  
cd build  
cmake ../src
```

4. So then did

```
cd ..
git clone https://github.com/sekigon-gonnoc/Pico-PIO-USB.git Pico-PIO-USB
git clone https://github.com/hathach/tinyusb tinyusb
```

5. Tried again:

```
cd build  
cmake ../src
```

- This gave an error:

```
CMake Error at CMakeLists.txt:25 (add_executable):
  Cannot find source file:

    /home/dev/pico/PicoGamepadConverter/src/device_files/rndis-lib/mongoose/mongoose.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc
```

6. I then tried dropping the mongoose.c and .h into the src - which did allow cmake to finish,  but then using make (to create the UF2) fails with:

```
[100%] Linking CXX executable PicoGamepadConverter.elf
/usr/lib/gcc/arm-none-eabi/12.2.0/../../../../arm-none-eabi/bin/ld: CMakeFiles/PicoGamepadConverter.dir/device_files/rndis-lib/mongoose/mongoose.c.obj: in function `p_mkdir':
mongoose.c:(.text.p_mkdir+0x6): undefined reference to `mkdir'
Memory region         Used Size  Region Size  %age Used
           FLASH:      498172 B         2 MB     23.75%
             RAM:       50032 B       256 KB     19.09%
       SCRATCH_X:          2 KB         4 KB     50.00%
       SCRATCH_Y:          0 GB         4 KB      0.00%
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/PicoGamepadConverter.dir/build.make:3271: PicoGamepadConverter.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:1678: CMakeFiles/PicoGamepadConverter.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

Loc15, 2024-04-16:
The problem is the fourth step, you need do **git submodule update --init** instead git cloning the submodules after the git clone, I use outdated version of mongoose.

andrewfstratton, 2024-04-16:
Thank you - I've added a pull request with build instructions that I've successfully tested on Win 11 with Docker Desktop

PR #7 (2024-04-18): Add instructions for building based on docker (Desktop for Windows)
This should Fix #6

I've tested this on Win 11 and it built correctly
closed 2024-04-18 by Loc15