Loc15/PicoGamepadConverter #8: Some Idea

Vitari-Translates, 2024-08-13:
I really love the project. I have some idea. Sometimes seems the pico is frozen. Add some led pulsating when the pico is booted properly. I love the web interface but if you use in a console its a bit circumstantial to switch other gamepad. I dont know is it posible or not but the 8bitdo adapter do it really good. When a gamepad is connected if you hold 2 spcific button the adapter change de output mode. For this need an rgb led to show the user which output mode is currently use. My last idea is the gamepad support. I know is it not possible to support all gamepad but the nintendont software is do it really good. You can create own cfg file for your controller. In the cfg you can associate the buttons.

Loc15, 2024-08-14:
> Sometimes seems the pico is frozen

At what moment happened that to you?

> but if you use in a console its a bit circumstantial to switch other gamepad

What modes do you use?

> For this need an rgb led to show the user which output mode is currently use

Always I have tried to use minimal additional hardware because the people don't want to spend on more things. I try keeping it easy. I have think about to add two fast mode to change between them, but maybe I keep using the original led for the feedback. 

>  I know is it not possible to support all gamepad but the nintendont software is do it really good. You can create own cfg file for your controller. In the cfg you can associate the buttons.

What gamepad do you use ? It should work on any gamepad (almost on theory). The difference between nintendont and this is nintendont not parse the info that you gamepad pass via HID descriptor, because that you need specificate the order of data of your gamepad. PicoGamepadConverter do this parse. I didn't testing enough gamepads because is difficult but it can read generic gamepads.

Vitari-Translates, 2024-08-14:
-It not frozen, i think just seems to frozen because dont have any feedback.

-I want to use Switch, Xinput, PS2, Dinput. And if i want to change the gamepad or the mode i need a PC to do it.
For the gamepad i dont have any idea how to do it. Maybe using MAC list instead of one MAC (if its possible).
For the mode change everything working fine, but need some feedback aka RGB led or just flashing the original led X times.

-I want to use a PowerA Switch Controller via bluetooth. In generic mode its works almost perfectly but the analog and triggers not.

Loc15, 2024-08-17:
> I want to use a PowerA Switch Controller via bluetooth. In generic mode its works almost perfectly but the analog and triggers not.

Yeah, it's not a generic controller, but I think someone might lend me one, and I'll try to fix it.