Loc15/PicoGamepadConverter #9: Emulation of a PS1 or PS2 controller.

retroconsolesgaming, 2024-08-18:
Hello
Thank you for the project. It's working well with ps4 pad on host and playstation pad simulation on my ps2, but I can't make it work with ps1. Is there any switch to change from working on ps2 to ps1?

Loc15, 2024-08-18:
Hey mate, I never tested on a real ps1 because I didn't have one, I guess I need to do some fixes then. Doesn't it work at all? Thanks for your feedback.

retroconsolesgaming, 2024-08-18:
I had more time now and I tested it more. Ps1 bios can't recognise pico controller at all. 
This time I tested more games.
- Fifa 2000(ps1):
On ps1 can recognise ps1 controller and ps2 controller, but can't see the pico controller
On ps2 is the same, ps1 and ps2 controller are working, but can't see the pico.
- Spyro the Dragon:
On ps1 is working great with pico controller, I even tried the ps5 controller and all is working.
- Croc Legend of the Gobbos
Pico controller not working

So it does working on ps1, but with some games and some not. 
There must be some difference on ps1 and ps2 controllers that pico is not working in ps1 bios and some games.
You did a great job with this project and thank you for this.

Loc15, 2024-08-19:
> On ps2 is the same, ps1 and ps2 controller are working, but can't see the pico.

Did you tested the ps1 game on ps2 ? Have you a fat ps2 ?

> On ps1 is working great with pico controller, I even tried the ps5 controller and all is working.

Nice.

> There must be some difference on ps1 and ps2 controllers that pico is not working in ps1 bios and some games.

The issue might be related to speed. I only tested on Ps2 which operate at a faster polling rate compared to the PS1. This difference in polling rate may affect how well the commands are received and processed.

retroconsolesgaming, 2024-08-19:
New tests. I checked few games from ps1 becouse they work both on ps1 and ps2.

- ps1 fat (I don't have a working laser for my fat, so I can only test in bios)
Not working in bios

- ps1 slim
Not working in bios
Not working in GTA2
Working in Spyro 3
Not working in fifa
Working in Medal of Honor

- ps2 fat
Working in bios
Not working in GTA2
Working in Spyro 3
Not working in Fifa
Working in Medal of Honor

- ps2 slim
Working in bios
Not working in GTA2
Working in Spyro 3
Not working in Fifa2000
Working in Medal of Honor

I've tried few ps2 games and all are working.
I tried all my controllers. The ps1 digital, dualshock 1 and dualshock 2 and they are working in every ps1 and ps2 console.

Loc15, 2024-08-21:
@retroconsolesgaming Thank you mate for your this info.

Loc15, 2024-09-08:
Hey mate, can you test these new fixes ? 
[github.zip](https://github.com/user-attachments/files/16921044/github.zip)

retroconsolesgaming, 2024-09-18:
Sorry for the delay. 
They are both working, but there is a problem with button mashing.
The second one is slightly better, but not perfect.

Loc15, 2024-09-18:
@retroconsolesgaming Hey mate, but you tested both consoles ?
What do you mean with the smashing ? is the response slow ?

retroconsolesgaming, 2024-09-18:
I was short on time, and tested only what didn't work previously. The ps1 bios, and fifa. On both version controller appear in bios and fifa. In fifa 2000 sprint button (triangle) must be pressed repeatedly for sprint. On 1 version it doesn't sprint at all, and on version 2 sprint works, but it has some breaks. I will try to find some program to test the controller, I have one app for ps2, but don't know if it can test ps1 controller.

Loc15, 2024-09-29:
I need to test with PS2 console again to know if these new versions works

ez2torta, 2024-10-30:
Hi! I was trying to connect the pico to the ps2 but i don't know how to power the pico. 
[image]

I followed the README file but it does not mention nothing about 3.3V and 5V.. how should i handle that?
Thanks

Loc15, 2024-10-30:
@ez2torta What host mode are you using ? If are using USB HOST mode like Xinput, Dinput then you need connect to VBUS and GND  via USB A connector to ps2's USB port. For the others modes then you can just use a normal USB cable connected to native pico's USB female port to ps2's USB port.

ez2torta, 2024-10-31:
> @ez2torta What host mode are you using ? If are using USB HOST mode like Xinput, Dinput then you need connect to VBUS and GND  via USB A connector to ps2's USB port. For the others modes then you can just use a normal USB cable connected to native pico's USB female port to ps2's USB port.

Omg it is working on the original pico! Thank you!

I tried a bootleg one but couldnt make it work.

Loc15, 2024-10-31:
@ez2torta 

> I tried a bootleg one but couldnt make it work.

What model it is ?

ez2torta, 2024-10-31:
It is this one

https://es.aliexpress.com/item/1005007090230795.html?spm=a2g0o.order_list.order_list_main.29.7a66194dEEBSnQ&gatewayAdapt=glo2esp

i tried using a usb c to usb A adapter but it did not work. 

Also i cleared KOF11 Arcade mode with this adapter so i'm super duper happy with this! i hope i can contribute to your project