Loc15/PicoGamepadConverter #12: How does Playstation Host option work?

Pazureka, 2024-09-30:
I think i am missing something important, i did connect PS2 cable to pi pico, and used PS3 controller after changing host mode to DINPUT via OTG cable, i also added power from another usb, but it appears totally dead.

Loc15, 2024-09-30:
You need do this [connection](https://github.com/Loc15/PicoGamepadConverter/blob/main/docs/ps1_pinout.png) and put a resistor between 3.3v and PS2 data pin

Pazureka, 2024-09-30:
Wouldnt it be for connecting playstation controller TO pico? I want to emulate playstation controller and use it with real PS2, either way, im gonna test now

Loc15, 2024-09-30:
I thought that you wanted host mode. If you want device mode use this [connection](https://github.com/Loc15/PicoGamepadConverter/blob/main/docs/PS1_PS2_pinout.png) instead and choose the device mode on the web configurator

Pazureka, 2024-09-30:
yeah, thats what i did, but nothing really happens when plugged into PS2.
[image]
I did quick and dirty visual representation of what i want to achieve so there would be no confusion.
Should arduino get power from battery powered controller? im not sure how its powered in other way.

Loc15, 2024-10-01:
[Do you see the additional component micro usb ?](https://github.com/Loc15/PicoGamepadConverter/raw/main/docs/wii_setup.jpg) I get power through this connector when native usb connector is on use. If you don't have one of these you can use another usb cable and just connect the positive to vbus and ground to ground and connect to USB of the Play2 console. Your diagram look good.

closed 2024-10-12 by Loc15