Loc15/PicoGamepadConverter #13: Cannot build on Ubuntu 24

ez2torta, 2024-10-31:
Hi!
I did a fork and wanted to build the project on my machine.

I'm getting an error while trying to build using make.

here's the output

`[  6%] Building C object CMakeFiles/PicoGamepadConverter.dir/utils.c.obj
/home/tortita/Coding/PicoGamepadConverter/src/utils.c:11:10: fatal error: pico/cyw43_arch.h: No such file or directory
   11 | #include "pico/cyw43_arch.h"
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/PicoGamepadConverter.dir/build.make:104: CMakeFiles/PicoGamepadConverter.dir/utils.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:1930: CMakeFiles/PicoGamepadConverter.dir/all] Error 2
make: *** [Makefile:91: all] Error 2`

any hints? do i need a lower version of the pico sdk?

Thanks!

Loc15, 2024-10-31:
You need Pico SDK 1.5.1

ez2torta, 2024-10-31:
Great!

now the build is working :) Thanks!

FYI: Anyone reading this, to lower the pico-sdk version i had to run on the pico-sdk folder

`git checkout 6a7db34ff63345a7badec79ebea3aaef1712f374`

closed 2024-10-31 by Loc15