Loc15/PicoGamepadConverter #14: PS2 Controller Laggy

ba114, 2024-12-14:
Hello,

Im using this with a ps2 controller. i want to be able to test controller inputs/

I have wired up as PS2 host mode, Xinput as the output. The controller is detected by windows and i can see it in the game controllers settings.

The issue is that all movements of the analog sticks are extremely laggy. Do you know why this is?

Loc15, 2024-12-14:
Did you put the pull up resistor on data pin ?  It's necessary for PS2 controller.

[Do the test with this](https://hardwaretester.com/gamepad)

ba114, 2024-12-14:
> Did you put the pull up resistor on data pin ? It's necessary for PS2 controller.
> 
> [Do the test with this](https://hardwaretester.com/gamepad)

Yes used 1k resistor on data line.
Results are same with the gamepad tester website,.

Loc15, 2024-12-15:
It's strange, I got that issue too when I didn't use the pull up resistor (I used a 10k resistor) but if you say that you used then I don't what is wrong.

What firmware version are you using?

ba114, 2024-12-15:
Latest version. I only used 1k resistor. I'll try again with 10k

Loc15, 2024-12-18:
I can't test again because I don't have a PS2 controller now. The last time I did test that feature was on 0.0.2 version