Loc15/PicoGamepadConverter #17: Layout issues with Razer Kishi v1

4ntsu, 2025-01-14:
Hello,

First and foremost, thank you for creating such a useful and versatile tool.

I'm trying to connect a Razer Kishi (v1) controller to a Pico W, with the main intention of using it as a Bluetooth controller.
The connection to the controller is successful with both Xinput and Dinput on the USB Host side, but the layout of the controller does not seem to be recognised properly. Below is a summary of what works and what doesn't:

Working: A, B, X, Y, Start, Select, D-pad

Not working:
- Left and right analog sticks' start and end values seem to be "warped" (see attached video).
- Left and right analog sticks' buttons do not register.
- Home button does not register.
- Shoulder buttons register as triggers.
- L and R triggers register as buttons.

Below is a video of how the analog sticks and triggers behave. I'm spinning the analog sticks clockwise (first left then right), then I press L1, R1, L2, L2:
<video src=https://github.com/user-attachments/assets/cb037fcb-f660-47a0-b1cf-8592f136429a />

I'm happy to test or provide any other info about the controller that may help.

Loc15, 2025-01-15:
Looks like[ Razer Kishi haven't a native Xinput mode ](https://www.pcgamingwiki.com/wiki/Controller:Razer_Kishi) otherwise it had worked on xinput mode. On Dinput mode it looks like I'm not parsing the right information from HID descriptor. The easier way is do a specific case for this controller because I don't have one to try myself. Looks too like you use linux, use [this](https://github.com/DIGImend/usbhid-dump) and dump the descriptor and through the streams identify the position of buttons.

4ntsu, 2025-01-15:
Hi, thanks for the quick reply. I have dumped the values below:
<details>
<summary>Razer Kishi v1 HID dump (click to expand)</summary>

```
Neutral: 00 00 00 00 80 00 00 00 00 00 00

R1: 00 00 00 00 80 00 80 00 00 00 00
L1: 00 00 00 00 80 00 40 00 00 00 00
R2: 00 00 00 00 80 00 00 02 00 FF 00 - The 8th byte (0x02) gets set to this value at any level. The 10th byte is the analog read from 0x00 to 0xFF when fully pressed
L2: 00 00 00 00 80 00 00 01 00 00 FF - The 8th byte (0x01) gets set to this value at any level. The 11th byte is the analog read from 0x00 to 0xFF when fully pressed

R3 (stick button):  00 00 00 00 80 00 00 40 00 00 00
L3 (stick button):  00 00 00 00 80 00 00 20 00 00 00

D-pad:
    Up:             00 00 00 00 00 00 00 00 00 00 00
    Up + Right:     00 00 00 00 10 00 00 00 00 00 00
    Right:          00 00 00 00 20 00 00 00 00 00 00
    Down + Right:   00 00 00 00 30 00 00 00 00 00 00
    Down:           00 00 00 00 40 00 00 00 00 00 00
    Down + Left:    00 00 00 00 50 00 00 00 00 00 00
    Left:           00 00 00 00 60 00 00 00 00 00 00
    Up + Left:      00 00 00 00 70 00 00 00 00 00 00
    
Select: 00 00 00 00 80 00 00 04 00 00 00
Start:  00 00 00 00 80 00 00 08 00 00 00
Home:   00 00 00 00 80 00 00 10 00 00 00

A: 00 00 00 00 80 00 01 00 00 00 00
B: 00 00 00 00 80 00 02 00 00 00 00
X: 00 00 00 00 80 00 08 00 00 00 00
Y: 00 00 00 00 80 00 10 00 00 00 00

Right stick:
    X Axis:
        All the way to the left:  00 00 81 00 80 00 00 00 00 00 00
        All the way to the right: 00 00 7F 00 80 00 00 00 00 00 00
    Y Axis:
        All the way up:     00 00 00 81 80 00 00 00 00 00 00
        All the way down:   00 00 00 7F 80 00 00 00 00 00 00

Left stick:
    X Axis:
        All the way to the left:  81 00 00 00 80 00 00 00 00 00 00
        All the way to the right: 7F 00 00 00 80 00 00 00 00 00 00
    Y Axis:
        All the way up:     00 81 00 00 80 00 00 00 00 00 00
        All the way down:   00 7F 00 00 80 00 00 00 00 00 00
```
</details>

For the sticks, the values increase from `0x01` to `0x7F` going right or down, and decrease from `0xFF` to `0x81` going left or up.
```                                     
  |-----------------------------------------------------------------------------|
  |                                 |   |   |                                   |
  |-----------------------------------------------------------------------------|
 0x81                             0xFF 0x00 0x01                              0x7F
```

Regarding Xinput/Dinput support, the controller is detected by PicoGamepadConverter with both host modes set in the web UI, so I assumed it supported Xinput too.

Thanks again!

Loc15, 2025-01-16:
> so I assumed it supported Xinput too

Nope, when Xinput mode it's selected it still detected HID devices, but the input will not readed.

I'm going use that information that you send to try to make a config but maybe I take a time.