Loc15/PicoGamepadConverter #18: add wii classic controller mode

EnzoMigliore, 2025-01-25:
sarebbe bello poter avere una modalita simile alla modalita gamecube ma che emuli il wii classic controller siccome gia ce la modalita che emula il wimote+nunchuk via bluethoot sarebbe bello avere anche la connessione cablate.

Loc15, 2025-01-25:
Wii classic controller already exist. [In order to change between mode press X and Y (Xbox layout) at same time ](https://github.com/Loc15/PicoGamepadConverter?tab=readme-ov-file#wii)

EnzoMigliore, 2025-01-25:
> Il controller classico Wii esiste già. [Per cambiare modalità, premere X e Y (layout Xbox) contemporaneamente](https://github.com/Loc15/PicoGamepadConverter?tab=readme-ov-file#wii)

yes but this mode does not work with the pico but only with the pico w via bluetooth

Loc15, 2025-01-25:
The Wiimotes works with Bluetooth and we are emulating one, the only way that emulate a Wiimote with not Bluetooth is via [software](https://github.com/xerpi/fakemote) modifying the Wii or [change the Wii's Bluetooth module](https://github.com/JSnowden33/Wii-Bluetooth-Replacement) and these are already different projects.

EnzoMigliore, 2025-01-25:
yes but the winnote connects via bluetooth instead the nunchuk connects via cable to the wiimote so wouldn't it be possible to have a mode that transforms the host controller for example the ps4 controller into a wii classic controller?

Loc15, 2025-01-26:
Yeah, it can be made on this way, I wanted to do this mode before, but now I don’t have time. Too pico W is almost the same price as a normal pico and use the pico W remove the needing to have a wiimote and buy batteries.