Loc15/PicoGamepadConverter #19: Adapter won't detect  Logitech F310

mariosilva555, 2025-02-02:
Hello

I Built one the adapter with the intent of connecting USB controllers on the PS2.

The adapter works fine with a GP2040-CE arcade stick, but it did not detect my Logitech F310

I tried to connect the controller on both Xinupt and Dinput modes on the adapter and the controller.

The F310 is working fine on a PC.

Loc15, 2025-02-02:
It's very weird because I have a F710 it works just fine on both modes and F310 and F710 are almost the same controller.
Are you doing the correct connection ? Do you choose the right mode on web config ? How you know it's not detected ?

mariosilva555, 2025-02-02:
> Are you doing the correct connection ?

I do believe it is correct. It is working fine with my arcade stick (GP2040-CE board).

>Do you choose the right mode on web config ?

Yes

>How you know it's not detected ?

I might have said that in the wrong way. When I plug the controller on the adapter, the indicator light turns on, but it will not register any command.

Loc15, 2025-02-03:
> When I plug the controller on the adapter, the indicator light turns on

Looks like it's detected. Has you tried others output modes ?

mariosilva555, 2025-02-04:
Hello. I just tested the Dinput output mode and it is working fine (Xinput on controller, connected as Dinput device on a computer).

I plugged the controller on the extra USB port, and the RPi on the computer. I am not sure if this issue could be a problem with my OTG adapter (I only use it when I plug the controller directly on the RPi).

I will try to get another adapter.

Loc15, 2025-02-04:
>  not sure if this issue could be a problem with my OTG adapter

Yeah try do another test with another adapter. Yesterday I did test again with my F710 and all was working fine.
Let me know how it goes.

mariosilva555, 2025-02-04:
Hello. Managed to borrow another OTG adapter, and I still have the issue. Works fine with my arcade stick, doesn't  work on the F310.

Loc15, 2025-02-07:
It's weird. What firmware version do you use ? What you get on UART output?

mariosilva555, 2025-02-09:
I tried 0.0.2, 0.0.3 and 0.0.4 alpha.

I don't know how to get a UART output. I tried to look for some information, but I could not find how.

If you could point me to some documentation, I will try to check it.

Loc15, 2025-02-13:
Connect GP0 (TX) to serial to USB converter or use another pico using picoprobe firmware

mariosilva555, 2025-02-19:
Hello. I managed to get my hands on a USB Serial converter.

Whens turning on the console with the F310, I get the following output:

XINPUT MODE HOST -> 0 | MODE DEVICE -> 4
Init psx simulator
XINPUT MOUNTED 01 0

I get the same thing on the GP2040CE arcade stick.

Loc15, 2025-02-27:
Looks like you gamepad it's working fine, I don't know why you don't getting the output

omersiar, 2025-12-26:
You mean forgetting it and repairing it again?  I tried that.