Loc15/PicoGamepadConverter #22: Crazy issues with PS1/PS2 controllers - Input delay and unexpected readings for analog sticks

boot-six, 2025-04-24:
I have wired up my PICO for PSX host mode, and it appears that the digital buttons are perfectly fine, but the analog sticks are wacky. On both my psOne revision of the dualshock and PS2 revision, I get different results. 

On the PSX Dualshock: 
- I have infrequent updates, seemingly only really reading inputs well if I spin the analog stick around the outer edge.
-  Finer movements still do not appear, and movements within reasonable distance do not appear to affect the input at all.

On PS2 Dualshock:
- Seemingly accurate input detection, but insane amounts of input delay. Like practically a full second before it reads input or reads that there is no output.

And yes, I do have a 10K resistor on DATA line.

Loc15, 2025-04-25:
Did you use the resistor when you tried with PS1 controller ? Try without it
What controllers models do you have ? See back of the controller.
What firmware version did you use ? Try with 0.0.2

> but insane amounts of input delay.

This happened to me too, but the pull-up resistor resolved it

boot-six, 2025-04-25:
> Did you use the resistor when you tried with PS1 controller ? Try without it. 
I tried it without it initially, on 0.0.4 firmware.
>What controllers models do you have? See back of the controller. 
SCPH-110 for PS1, SCPH-10010 for PS2.
>What firmware version did you use ? Try with 0.0.2
I tried 0.0.4 with and without the pullup resistor, and 0.0.2 only with the resistor. Both acted the exact same.

Loc15, 2025-04-25:
It's very weird, I did the test with a SCPH-110 and it worked just fine.

> infrequent updates

Your output freezes, or you say that because you need to spin the analog stick around the outer edge  ?

https://github.com/user-attachments/assets/626ef380-76ac-4d50-8c90-f584f988302c

boot-six, 2025-04-25:
That’s really odd. Is that video taken on 0.0.2 or 0.0.4 with or without the resistor? I do believe everything is wired up correctly as it does function just not well. The only other thing I could think of is I didn’t solder directly to the controller pins and I’m using a female PSX port, but I can’t see that causing that much of an issue.

Loc15, 2025-04-25:
>  Is that video taken on 0.0.2

 Yes

> with or without the resistor?

No resistor

>  didn’t solder directly to the controller pins

I didn't either, I did use male to female dupont jumpers

I think I need to test again on 0.0.4

boot-six, 2025-04-25:
Okay, I’ll remove my resistor and try 0.0.2 later to see what I get and report back. I’ll also test it on my actual PS1 to see if it’s acting weird there. It wasn’t the last time I used it but maybe something has changed.

Would you mind linking some of those DuPont pins? Just want to grab some in case that is the issue.

Loc15, 2025-04-25:
> Okay, I’ll remove my resistor and try 0.0.2 later to see what I get and report back.

Okay thanks

> Actually I just googled them and I think they’re the exact same pins in the female port I have. I extracted one and they look the same.

Yeah, I didn't think that was the issues

closed 2025-04-25 by boot-six

boot-six, 2025-04-25:
I removed my resistor, connected the data line direct, and flashed the 0.0.2 firmware to my Pico, but the issue still persists. For giggles I changed from Xinput device to Switch device to see if that would help, as well as changed which USB port it was in, and I am still getting the same issue. I also decided to try it on my Switch while I had it in Switch device mode and the issue persists there, as well. 

I have triple checked that all my connections are correct and that all my pins are contacting, but still no luck.

Loc15, 2025-04-26:
Can you record a video moving the sticks using https://hardwaretester.com/gamepad on xinput mode ? Thanks

boot-six, 2025-04-26:
https://github.com/user-attachments/assets/bf395aee-6da0-4102-b061-6e54d9c0ad58

Loc15, 2025-04-26:
Nice. The choppy move I think it's normal because I didn't adjust right the value of the stick when I pass the data to 8 bits to 16 bits. It's something I have to fix.
What really I did worry was the 

> insane amounts of input delay

on ps2 controller

What about the ps2 controller ? it worked ? (On this case it need the resistor)

boot-six, 2025-04-26:
https://github.com/user-attachments/assets/8010992e-1a6c-42b9-9635-48b408fa416e

I had to encode this to a lower quality and frame rate to allow github to upload, so I apologize for that. This is my example of a Dualshock 2 with a 10k resistor on Data line, running 0.0.2. For reference, there is no deadzone in my config, and deadzones in game are set to None. 

The first couple turns I am bringing my stick to the left and right edges attempting to swing my camera back and forth on the sides of the rock structure. I then attempt to follow the moving target with constant fine movements, and then I attempt to flick back and forth onto the target. These are all movements I can do consistently on a normal Xbox series controller, but here the input is noticeably wrong even by human reaction measurements.

Loc15, 2025-04-29:
Hey mate, I borrowed a friend's ps2 controller and I can did the test again. It's what you tell me, but I did the test with a ps2 controller copy. The sticks data is like it had poor resolution and I don't know if it's because the controller it's not original. Your controller is legit or a copy ?

I did the test with others software [1](https://github.com/SukkoPera/PsxNewLib) [2](https://github.com/todbot/CircuitPython_PS2Controller) and I get the same result

boot-six, 2025-04-29:
My controller is an original DualShock 2, yes. And I agree, it appears that both just have a difficult time keeping up with changes to the analog data.

Loc15, 2025-05-06:
Yeah, It's then something I need to fix

Loc15, 2025-10-26:
@Toad86  
> It doesn't do it with version H. Also 1200 A, H and M are okay.

Did you use a pull resistor ?

Thank you for the info