Loc15/PicoGamepadConverter #23: Xbox 360 Support

sanjay900, 2025-05-09:
We found the keys for these consoles, so if you ever want to support actual 360 consoles you can now.
https://github.com/InvoxiPlayGames/libxsm3
https://github.com/Santroller/Santroller

Loc15, 2025-05-10:
Thank you, it's great. I guess still need a extended example how implementing this.

sanjay900, 2025-05-10:
First thing youd need to do is swap out your current xinput usb descriptor for a full one with all the interfaces.

Then its as simple as copying libxsm3 into your project, and when you get the control requests for auth, you just hand them off to libxsm3.

My implementation of this is in here: https://github.com/Santroller/Santroller/blob/master/src/shared/usb/descriptors.cpp

```
switch (request) {
        case 0x81:
            xsm3_initialise_state();
            xsm3_set_identification_data(xsm3_id_data_ms_controller);
            memcpy(requestBuffer, xsm3_id_data_ms_controller, sizeof(xsm3_id_data_ms_controller));
            return sizeof(xsm3_id_data_ms_controller);
        case 0x82:
            xsm3_do_challenge_init((uint8_t *)requestBuffer);
            return 0;
        case 0x87:
            xsm3_do_challenge_verify((uint8_t *)requestBuffer);
            return 0;
        case 0x84:
            return 0;
        case 0x83:
            memcpy(requestBuffer, xsm3_challenge_response, sizeof(xsm3_challenge_response));
            return wLength;
        case 0x86:
            short state = 2;  // 1 = in-progress, 2 = complete
            memcpy(requestBuffer, &state, sizeof(state));
            return sizeof(state);
    }
```

Loc15, 2025-05-10:
Okay I'm gonna try it and let you know.

Loc15, 2025-11-30:
@Invictaz I don't have a xbox 360 and I don't know anyone that have one, so it's difficult to start to implement without actual hardware to test