Loc15/PicoGamepadConverter #25: Is there a way to remap or bind the keys?

Ambassaman, 2025-07-11:
I'm truly impressed by this project.

However, one thing I find a bit lacking is that it doesn't seem to allow remapping of button inputs.

I’d like to use a PSX Densha de Go controller on actual consoles, and with the right button assignments, it works perfectly on systems like the Saturn, Nintendo 64, and Dreamcast.

While there are many adapters that convert XInput or DInput for use on consoles, very few of them offer the option to customize button mappings—which is unfortunate. I need to remap all digital buttons, including the D-pad.

If you could provide this feature, I believe your device could be extremely useful in many other applications as well. From what I know, there are quite a few devices that rely on custom button mappings to function properly.

Thank you.

Loc15, 2025-07-13:
Hi, actually not is possible to remap easily the button input. You going to need to edit the source files and compile.

It's planned to add Nintendo 64 support. Dreamcast, for other hand, I don't know because it's not an easy console to get.

Remapping it's not currently a priority task, but it maybe can be added in the future.

fixongbg, 2025-08-14:
@Ambassaman I came to this GitHub for the exact same reason. Getting the Densha de Go PS1 Taito controller to work with the Switch. 

Any luck getting it to work in analog or step mode with the picogamepadconverter at all?